ycheng517/ar4_ros_driver #2: Getting the gripper working on AR4-MK2

se1exin, 2024-04-07:
Hi @ycheng517, thanks for your efforts in adding the gripper! I just thought I'd post my findings when testing this on the AR4-MK2

First, it was not working at all for but I eventually figured out I had to install `ros-iron-gripper-controllers` using apt.

```
sudo apt install ros-iron-gripper-controllers
```

Second, the gripper moves in reverse - that is, the `closed` state in MoveIt actually opens the gipper, and vice-versa. I'm not sure if this is related to my servo though - it is a HITEC HS-311.

To fix this I just inverted the values in `ar_gripper_macro.sdrf.xacro`.
 
See https://github.com/se1exin/ar4_ros_driver/commit/aee4e189507db1bc71a552c59ab89b9721278384

se1exin, 2024-04-07:
An update to the gripper direction issue. I've replaced my servo with an MG995 and now the original direction is correct.

The gearbox in the HITEC HS-311 must in the opposite direction...

ycheng517, 2024-04-10:
Thanks for pointing out the need to install `ros-iron-gripper-controllers`, I've added `gripper_controllers` as a dependency in de761b1954506996d9f08a6ce815fa24f2a04460

Regarding the gripper direction issue, glad you were able to figure it out!

closed 2024-04-10 by ycheng517