ycheng517/ar4_ros_driver #3: AR4-MK2 in wrong position after calibration

se1exin, 2024-04-24:
After starting `ar_hardware_interface` with `ros2 launch ar_hardware_interface ar_hardware.launch.py calibrate:=True` I've found that J4 and J5 are rotated too far by around ~10 degrees or so.

This is compared to calibrating the arm using the Annin Robotic's `Teensy 4.1 sketch Ver 4.3` and calibrating using the `AR4 Version 4.3.1` software, which ends up with J4 being correct and J5 only slight off due to the position of the limit switch on my arm.

I've already tried to debug this myself by modifying `REST_ENC_POSITIONS` and `JOINT_LIMIT_MIN` and `JOINT_LIMIT_MAX` but have not had any success. I'm happy to debug this further if you're able to give me some hints on what else to try modifying in the Arduino sketch? 

Image of arm after calibration using Annin Robotics firmware and software:
[image]

Image of arm after calibration using `ar_hardware_interface`:
[image]

ycheng517, 2024-04-25:
You need to change `joint_offsets_` here: https://github.com/ycheng517/ar4_ros_driver/blob/main/ar_hardware_interface/src/ar_hardware_interface.cpp#L37. Btw, I'll move this to a YAML config soon.

You can optionally change `REST_ENC_POSITIONS` to something more desirable. There should be no need to change `JOINT_LIMIT_MAX` and `JOINT_LIMIT_MIN`. 

In RViz, go to the "home" position (which should have all joints at zero degrees) to verify that the joints are aligned.

ycheng517, 2024-04-26:
I just updated the code, new location to update joint offsets is at: https://github.com/ycheng517/ar4_ros_driver/blob/main/ar_hardware_interface/config/joint_offsets.yaml

se1exin, 2024-04-26:
Love your work @ycheng517 !

The new `joint_offsets.yaml` made it really easy to home it in, thanks! My offset value for J5 is `-76`, which I think means it is 29deg out?

> For J5 because the offset is so large, it could be an issue that the encoder resolution is different for that joint. To see if that's the issue, verify that moving 10 degrees in RViz really results in moving 10 degrees in real life.

I did multiple measurements to check the angle of J5, and at small movements (e.g. 10deg), it appears to be about 3 degrees out. However, when doing a 30 degree rotation that error is reduced to around 1 degree. My robot is 3D printed so I'm going to chalk that up to mechanical factors.

I do wonder if this error margin is playing a part in the initial calibration, but since I'm able to account for it using the `joint_offsets` file, and I now understand a bit more about the shortcomings of J5 on my arm when doing small movements, I think I'm happy with the calibration setup :)

Thanks again!

closed 2024-04-26 by se1exin