ycheng517/ar4_ros_driver #5: Wrong yaml reading

newtondotcom, 2024-04-25:
`[rviz2-2] [ERROR] [1714055338.561250897] [moveit_background_processing.background_processing]: Exception caught while processing action 'loadRobotModel': parameter 'robot_description_planning.joint_limits.joint_1.max_velocity' has invalid type: Wrong parameter type, parameter {robot_description_planning.joint_limits.joint_1.max_velocity} is of type {double}, setting it to {string} is not allowed. ` 

The file concerned is `ar_moveit_config/config/joint_limits.yaml`, i tried changing the order of joints declaration but there is always an issue with the join1 max velocity.

ycheng517, 2024-04-26:
After googling this error, you seem to have the same issue as https://github.com/ros-planning/moveit2/issues/1049, see this for the fix https://github.com/ros-planning/moveit2/issues/1049#issuecomment-1047029751

newtondotcom, 2024-04-26:
Thanks for the quick response, i will try later today using `LC_NUMERIC=en_US.UTF-8` in front of my command ... I will keep you posted, thanks again

newtondotcom, 2024-04-26:
It worked, thanks you so much ! Loving your incredible work ! Keep up 🔥

closed 2024-04-26 by newtondotcom