ycheng517/ar4_ros_driver #7: Way to control arm and gripper externally

newtondotcom, 2024-04-28:
Hey, I was wondering if there is a clean way to control the arm and gripper externally than with rivz moveit extension ? 

For the arm, I managed to do so using a publisher in a `.py` : `create_publisher(JointTrajectory, '/joint_trajectory_controller/joint_trajectory', 10)` but for example, I don't know if the trajectory went well, etc ..

For the gripper, i don't see any topic regarding `gripper_cmd` so I didn't managed to make it work.

Do you have any clue ?

BR

newtondotcom, 2024-04-29:
In fact, it seems that `moveit`doesn't publish its namespaces so we can't use : `FollowJointTrajectory` and `GripperCommand` as topics.

newtondotcom, 2024-04-29:
okay my bad, i was searching for a topic, but it is an action. So sorry !

closed 2024-04-29 by newtondotcom

ycheng517, 2024-04-30:
See this script for an example of using `MoveItPy` to control the arm: https://github.com/ycheng517/ar4_ros_driver/blob/main/ar_hand_eye/scripts/follow_aruco_marker.py

I haven't looked into gripper control yet. 

If you have a working solution that directly publishes to the right topics, drop a link here! I'd be interested to know how to do that as well.

se1exin, 2024-05-17:
Hi @newtondotcom I've been experimenting with moving the arm & gripper externally and have some working code in a test repo: https://github.com/se1exin/ros2_ar4

If you look at https://github.com/se1exin/ros2_ar4/blob/master/src/ar_testing/ar_testing/ar_moveit_test.py you can see some demo functions for controlling the arm and the gripper using MoveItPy.

The readme has instructions for running the ar_moveit_test file.

Additionally, I've been playing with controlling the arm using voice control. If you look in the `ar_whisper` package (https://github.com/se1exin/ros2_ar4/blob/master/src/ar_whisper/ar_whisper/ar_whisper.py) you can see how I control the robot arm using `ros2_control` and manipulating the joint states directly.

Keep in mind that repo is just my test code/playground. It's full of commented out code etc, so it's pretty messy, but you're welcome to have a poke around if it helps you get your arm moving!