ycheng517/ar4_ros_driver #8: MotionPlanning warning in Rviz

EbrahimAlmajed, 2024-05-01:
Hello I'm new to ros2 and rvis and i wanted to test AR4 in gazebo and when i tried to use MotionPlanning i get this warning and a plan error.
[image]

Not sure if this is related but i use `rosdep install --from-paths . --ignore-src -r -y`
I get the following error
```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
ar_hardware_interface: Cannot locate rosdep definition for [ros2_aruco]
ar_hand_eye: Cannot locate rosdep definition for [ros2_aruco_interfaces]
```
using ros2 iron on ubuntu 22.04

ycheng517, 2024-05-02:
Ah sorry, this is because the external repos needed for hand-eye calibration is missing. Try running `vcs import . --input ar_hand_eye/hand_eye_calibration.repos`. Will update the instructions to include this step.

EbrahimAlmajed, 2024-05-02:
Thank you for replying, unfortunately now I get this error in Rviz 
`[ERROR] [move_group-1]: process has died [pid 54326, exit code -6, cmd '/home/user/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group --ros-args --params-file /tmp/launch_params_ezlnto9e --params-file /tmp/launch_params__m708mf8 --params-file /tmp/launch_params_73ugkyn4 --params-file /tmp/launch_params_x9zh94vk --params-file /tmp/launch_params_1sw2fyya --params-file /tmp/launch_params_13bwn6j1 --params-file /tmp/launch_params_3mizdyug --params-file /tmp/launch_params_51v3bk_o --params-file /tmp/launch_params_0xej9lo1'].`

this is the full log when running Rviz
```
[INFO] [launch]: Default logging verbosity is set to INFO
[INFO] [move_group-1]: process started with pid [54695]
[INFO] [rviz2-2]: process started with pid [54697]
[move_group-1] [WARN] [1714671376.411488260] [move_group]: Falling back to using the the move_group node namespace (deprecated behavior).
[move_group-1] [INFO] [1714671376.433386905] [move_group.moveit.moveit.ros.rdf_loader]: Loaded robot model in 0 seconds
[move_group-1] [INFO] [1714671376.433424576] [move_group.moveit.moveit.core.robot_model]: Loading robot model 'ar'...
[move_group-1] [INFO] [1714671376.433438716] [move_group.moveit.moveit.core.robot_model]: No root/virtual joint specified in SRDF. Assuming fixed joint
[move_group-1] [INFO] [1714671376.479300615] [move_group.moveit.moveit.kinematics.kdl_kinematics_plugin]: Joint weights for group 'ar_manipulator': 1 1 1 1 1 1
[move_group-1] [INFO] [1714671376.618743989] [move_group.moveit.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1714671376.618838940] [move_group.moveit.moveit.ros.moveit_cpp]: Listening to 'joint_states' for joint states
[move_group-1] [INFO] [1714671376.619307859] [move_group.moveit.moveit.ros.current_state_monitor]: Listening to joint states on topic 'joint_states'
[move_group-1] [INFO] [1714671376.619539883] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to '/attached_collision_object' for attached collision objects
[move_group-1] [INFO] [1714671376.619551524] [move_group.moveit.moveit.ros.planning_scene_monitor]: Stopping existing planning scene publisher.
[move_group-1] [INFO] [1714671376.619619405] [move_group.moveit.moveit.ros.planning_scene_monitor]: Stopped publishing maintained planning scene.
[move_group-1] [INFO] [1714671376.619766218] [move_group.moveit.moveit.ros.planning_scene_monitor]: Publishing maintained planning scene on 'monitored_planning_scene'
[move_group-1] [INFO] [1714671376.619804588] [move_group.moveit.moveit.ros.planning_scene_monitor]: Starting planning scene monitor
[move_group-1] [INFO] [1714671376.620182326] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to '/planning_scene'
[move_group-1] [INFO] [1714671376.620193166] [move_group.moveit.moveit.ros.planning_scene_monitor]: Starting world geometry update monitor for collision objects, attached objects, octomap updates.
[move_group-1] [INFO] [1714671376.620367729] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to 'collision_object'
[move_group-1] [INFO] [1714671376.620550932] [move_group.moveit.moveit.ros.planning_scene_monitor]: Listening to 'planning_scene_world' for planning scene world geometry
[move_group-1] [WARN] [1714671376.621857017] [move_group.moveit.moveit.ros.occupancy_map_monitor]: Resolution not specified for Octomap. Assuming resolution = 0.1 instead
[move_group-1] [ERROR] [1714671376.621870227] [move_group.moveit.moveit.ros.occupancy_map_monitor]: No 3D sensor plugin(s) defined for octomap updates
[move_group-1] terminate called after throwing an instance of 'rclcpp::ParameterTypeException'
[move_group-1]   what():  expected [string_array] got [string]
[move_group-1] Stack trace (most recent call last):
[move_group-1] #17   Object "", at 0xffffffffffffffff, in 
[move_group-1] #16   Object "/home/user/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x58db53968204, in _start
[move_group-1] #15   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7dd6ff029e3f, in __libc_start_main
[move_group-1] #14   Object "/usr/lib/x86_64-linux-gnu/libc.so.6", at 0x7dd6ff029d8f, in 
[move_group-1] #13   Object "/home/user/ws_moveit/install/moveit_ros_move_group/lib/moveit_ros_move_group/move_group", at 0x58db539671c3, in main
[move_group-1] #12   Object "/home/user/ws_moveit/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7dd6fffdb6a7, in moveit_cpp::MoveItCpp::MoveItCpp(std::shared_ptr<rclcpp::Node> const&, moveit_cpp::MoveItCpp::Options const&)
[move_group-1] #11   Object "/home/user/ws_moveit/install/moveit_ros_planning/lib/libmoveit_cpp.so.2.9.0", at 0x7dd6fffda5cb, in moveit_cpp::MoveItCpp::loadPlanningPipelines(moveit_cpp::MoveItCpp::PlanningPipelineOptions const&)
[move_group-1] #10   Object "/home/user/ws_moveit/install/moveit_ros_planning/lib/libmoveit_planning_pipeline_interfaces.so.2.9.0", at 0x7dd6ffa7efab, in moveit::planning_pipeline_interfaces::createPlanningPipelineMap(std::vector<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> >, std::allocator<std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > > > const&, std::shared_ptr<moveit::core::RobotModel const> const&, std::shared_ptr<rclcpp::Node> const&, std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char>  [... cut ...]
[rviz2-2] [INFO] [1714671380.337821537] [interactive_marker_display_97822652761904]: Service response received for initialization
[rviz2-2] [INFO] [1714671385.296156137] [rviz2_moveit.moveit.ros.planning_scene_monitor]: Failed to call service get_planning_scene, have you launched move_group or called psm.providePlanningSceneService()?
[rviz2-2] [INFO] [1714671385.296257859] [rviz2_moveit.moveit.ros.motion_planning_frame]: group ar_manipulator
[rviz2-2] [INFO] [1714671385.296268339] [rviz2_moveit.moveit.ros.motion_planning_frame]: Constructing new MoveGroup connection for group 'ar_manipulator' in namespace ''
```

ycheng517, 2024-05-03:
I feel your moveit packages are out-of-date. Try updating your packages or ROS distro version, or run in docker instead. The follow commands should work for sure.

```
docker build -t ar4_ros_driver .
./run_in_docker.sh
colon build
source install/setup.bash
ros2 launch ar_moveit_config demo.launch.py
```

ycheng517, 2024-05-03:
You can also try: https://github.com/ycheng517/ar4_ros_driver/issues/5

Not the same error but could be related.

EbrahimAlmajed, 2024-05-03:
I tried the docker container it work when using perfectly 
`ros2 launch ar_moveit_config demo.launch.py`
but had some issues when trying to simulate in gazebo using
`ros2 launch ar_gazebo ar_gazebo.launch.py`
and I tried #5 with no results but you are right its most likely a moveit or ros2 installation issue I will reinstall it and see if that makes any difference.

ycheng517, 2024-05-04:
gazebo demo should be working, feel free to send me the error you have

EbrahimAlmajed, 2024-05-07:
After trying again with Rviz and Gazebo with the docker container it worked perfectly. I still have the local issue probably an insulation issue on my part. Thank you so much for your support.

closed 2024-05-07 by ycheng517