ycheng517/ar4_ros_driver #9: Unable to successfully run ar_hand_eye calibrate.launch.py

se1exin, 2024-05-21:
Hi there, I'm having issues running the `ar_hand_eye` package. Rviz is loading, as well as my camera, but when the marker is is in from of the camera the calibration GUI does not show.

I think I've tracked it down to the following logs:

```
[rqt_calibrator.py-8] Warning: class_loader.ClassLoader: SEVERE WARNING!!! Attempting to unload library while objects created by this loader exist in the heap! You should delete your objects before attempting to unload the library or destroying the ClassLoader. The library will NOT be unloaded.
[rqt_calibrator.py-8]          at line 127 in ./src/class_loader.cpp
[ERROR] [rqt_calibrator.py-8]: process has died [pid 55247, exit code 1, cmd '/home/surface/dev/ROBOT_ARM/ar4_ros_driver/install/easy_handeye2/lib/easy_handeye2/rqt_calibrator.py --ros-args --log-level debug --ros-args -r __node:=handeye_rqt_calibrator --params-file /tmp/launch_params_20tc3zld'].
```
Looking closer at `rqt_calibrator.py` - it appears to be calling the `RqtHandeyeCalibrator` PythonQt widget.
When opening `RqtHandeyeEvaluatorWidget` in vscode, the editor complains that some of the `python_qt_binding` imports cannot be found. E.g.:
```
Import "python_qt_binding.QtCore" could not be resolved
```

I've install `ros-iron-python-qt-binding`, `PyQt5`, and even tried `pyside2`, but with no luck.

Do you know if there's any other dependencies I might need to install? Or am I looking in the wrong direction all together?

Additionally, the following logs are _sometimes_ shown when running the calibration launch file:
```
[handeye_server-7] [ERROR] [1716278208.692723726] [handeye_server]: The specified tf frames for the tracking system base/camera and marker do not seem to be connected
[handeye_server-7] [ERROR] [1716278208.693496974] [handeye_server]: Run the following command and check its output:
[handeye_server-7] [ERROR] [1716278208.694265647] [handeye_server]: ros2 run tf2_ros tf2_echo camera_link calibration_aruco
[handeye_server-7] [ERROR] [1716278208.694875688] [handeye_server]: You may need to correct the base_frame or effector_frame argument passed to the easy_handeye2 launch file
[handeye_server-7] [ERROR] [1716278208.695449724] [handeye_server]: Underlying tf exception: "calibration_aruco" passed to lookupTransform argument source_frame does not exist. 
[handeye_server-7] [WARN] [1716278208.696346813] [handeye_server]: Waiting for TF initialization...
[handeye_server-7] [INFO] [1716278208.698935908] [handeye_server]: Checking that the expected transforms are available in tf
[handeye_server-7] [INFO] [1716278208.699727565] [handeye_server]: Robot transform: base_link -> link_6
[handeye_server-7] [INFO] [1716278208.700455301] [handeye_server]: Tracking transform: camera_link -> calibration_aruco
[handeye_server-7] [INFO] [1716278211.770035636] [handeye_server]: All expected transforms are available on tf; ready to take samples
```

se1exin, 2024-05-21:
An update, when I run `rqt_calibrator.py` directly in a terminal and then run `calibrate.launch.py` in another the GUI shows when detecting the marker:

```
ros2 run easy_handeye2 rqt_calibrator.py --ros-args \
  -p name:=test \
  -p calibration_type:=eye_on_base \
  -p robot_base_frame:=base_link \
  -p robot_effector_frame:=link_6 \
  -p tracking_base_frame:=camera_link \
  -p tracking_marker_frame:=calibration_aruco
```
```
ros2 launch ar_hand_eye calibrate.launch.py
```

Any ideas why it is still failing when run from the launch file?

ycheng517, 2024-05-21:
these are the dependencies needed by easy_handeye2, have you installed them? https://github.com/marcoesposito1988/easy_handeye2/blob/master/easy_handeye2/package.xml#L27-L29

The install command would be:
```bash
rosdep install --from-paths . --ignore-src -y
```

> Additionally, the following logs are sometimes shown when running the calibration launch file:

These are supposed to show up on every run, after the aruco marker is detected. After these lines, the GUI starts up

se1exin, 2024-05-24:
Thanks for your response @ycheng517 

I had run `rosdep install --from-paths . --ignore-src -y` a few times to make sure the deps were installed, but after looking through those dependencies you linked I found I also had to install `ros2-iron-rqt`. Strange that the plugins where installed but the base package was not. Maybe the `package.xml` needs to be updated also include that package?

Either way, after installing `ros2-iron-rqt` with `apt` the launch file now works and the GUI shows when detecting the marker.

closed 2024-05-24 by se1exin

se1exin, 2024-05-24:
I got it working - thanks again for your help. Here's a video of it in action, the calibration is a bit off though haha.

https://www.youtube.com/shorts/LCqsrjWtK4I

I've ordered a realsense cam, looking forward to trying your tabletop-handybot repo out when it arrives!

ycheng517, 2024-05-27:
awesome, glad you figured it out! Looking forward to seeing more demos from you!