ycheng517/ar4_ros_driver #19: Gripper servo position configuration?

navdotnetreqs, 2024-08-16:
Hi!

So far I've been unable to find where the actual position sent to the gripper control for "open" and "closed" position comes from. Can these be modified? I have only found values "10" and "30" in run_arduino_nano.cpp, but assuming these are just a test, Chris' example in the manual is 0 and 60 degrees.

Br, J

navdotnetreqs, 2024-08-17:
By playing around with the values in ar_gripper_macro.srdf.xacro (open&closed) and recompiling, one can adjust the gripper positions.

navdotnetreqs, 2024-09-17:
The guy responsible is found in "ar_servo_gripper_hw_interface.hpp":

  int linear_to_angular_pos(double linear_pos) {
    return static_cast<int>(asin(-linear_pos / servo_arm_length_) * 180 / M_PI +
                            zero_deg_offset_);
  };
  
  This converts the given distance (the gripper joint is prismatic) to degrees.

closed 2024-09-17 by navdotnetreqs

ycheng517, 2024-09-23:
Though that works, here's the better place: https://github.com/ycheng517/ar4_ros_driver/blob/799dc994c1f54cfdaee946e2a8d1999de90f9afa/ar_moveit_config/srdf/ar_gripper_macro.srdf.xacro#L16

ycheng517, 2024-09-23:
Thanks for reporting the place to configure this, you're correct!