ycheng517/ar4_ros_driver #34: New planner format?

oskarsVismanis, 2025-01-14:
According to this [PR](https://github.com/moveit/moveit2/pull/2457), the format of how the controllers are loaded has changed in MoveIt2 and, I assume, to make this package compatible with Jazzy, @ycheng517, you've updated the format as well? I noticed this when running `demo.launch.py` and getting the error `expected [string_array] got [string]`. 

And besides this, using the `request_adapters` specified in the current `ompl_planning.yaml`
```
  - default_planning_request_adapters/CheckStartStateBounds
  - default_planning_request_adapters/CheckStartStateCollision
  - default_planning_request_adapters/ResolveConstraintFrames
  - default_planning_request_adapters/ValidateWorkspaceBounds
```
I get the error that these do not exist, but the demo works fine with these ones specified
```
default_planner_request_adapters/AddTimeOptimalParameterization
default_planner_request_adapters/FixWorkspaceBounds
default_planner_request_adapters/FixStartStateBounds
default_planner_request_adapters/FixStartStateCollision
default_planner_request_adapters/FixStartStatePathConstraints
```
I assume this is all from the move to moveit2 jazzy branch. As I'm still on humble, I fixed all this, so it would work for me, but perhaps a `humble` branch should be created? Or is that function currently done by the `iron` branch? As there is now the `annin_` prefix to packages, I was thinking a new branch would make more sense?

I am also working on adding a `prefix` or `tf_prefix` to the `.urdf`-s, which I could then contribute to the package (probably to the `iron`/`humble` branch initially, as that is the one I'm working with)

ycheng517, 2025-01-15:
Hi @oskarsVismanis I will make a `humble` branch in a short while. It'll take some effort to get Gazebo working again, as it was a big change to upgrade to Jazzy. Anyways stay tuned!

ycheng517, 2025-01-21:
I just created a fully working and updated humble branch at: https://github.com/ycheng517/ar4_ros_driver/tree/humble

oskarsVismanis, 2025-01-21:
> I just created a fully working and updated humble branch at: https://github.com/ycheng517/ar4_ros_driver/tree/humble

At the moment it seems `demo.launch.py` and `moveit.launch.py` doesn't launch the controllers correctly, as it defaults to CHOMP. I will look into it! 

Any reason for taking out pilz and switching to ignition gazebo instead of the previously used?

ycheng517, 2025-01-21:
I was able to get both of those launch files working in docker. This repo doesn't use CHOMP. Is that one of your custom changes?

I took out PILZ because it wasn't working (I wasn't able to click the dropdown of planners in Rviz and get the list of controller configs) and I didn't bother with fixing it since I'm not sure who actually uses PILZ. I may try to fix it when I have time.

I switched to Gazebo Ignition because 1. it was easier since Jazzy branch is already on it. 2. It will be easier to backport changes from the Jazzy branch to the humble branch in the future, if the code is similar.

oskarsVismanis, 2025-01-21:
When launching `demo.launch.py`, while loading ompl, the following messages are printed:
`
[move_group-3] [INFO] [1737470173.024806483] [moveit.ros_planning_interface.moveit_cpp]: Loading planning pipeline 'ompl'
[move_group-3] [INFO] [1737470173.031885207] [moveit.ros_planning.planning_pipeline]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'chomp_interface/CHOMPPlanner' for now.
[move_group-3] [INFO] [1737470173.034824667] [moveit.ros_planning.planning_pipeline]: Using planning interface 'CHOMP'
`

It was like this when cloning the branch, I will compare with the iron branch and see what might be doing this!

About PILZ, it was working for me in the iron branch, but the fact it's not really used is valid 😆 

And I was asking about Gazebo, because I was under the impression that ignition is no longer used in newer ROS versions (as I understood it's back to just Gazebo), but I was probably wrong so no worries, thanks!

ycheng517, 2025-01-22:
I do see the The `Multiple planning plugins available. You should specify the '~planning_plugin' parameter` problem as well. Though I default to OMPL. 

```
[move_group-1] [INFO] [1737552930.302890898] [moveit.ros_planning.planning_pipeline]: Multiple planning plugins available. You should specify the '~planning_plugin' parameter. Using 'ompl_interface/OMPLPlanner' for now.
[move_group-1] [INFO] [1737552930.308573740] [moveit.ros_planning.planning_pipeline]: Using planning interface 'OMPL'
```

It should be fixed with https://github.com/ycheng517/ar4_ros_driver/commit/1e0b20336717d3d85d539f4ff5e43552d0ff7e09. Hopefully with that, your moveit will default to using OMPL now.

oskarsVismanis, 2025-01-23:
Yup, the change fixed it, thanks! Closing this issue as well, as it's no longer relevant!

closed 2025-01-23 by oskarsVismanis