ycheng517/ar4_ros_driver #39: package failed: annin_ar4_description

alonborn, 2025-02-02:
Hi,
first - wow! it looks great! thank you for developing this excellent tool!
I followed your installation instruction and for some reason, it fails when trying to install annin_ar4_description.
Does the following tell you something? any idea how to resolve this?

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies (ROS distro is not set. Make sure `ROS_DISTRO` environment variable is set, or use `--rosdistro` option to specify the distro, e.g. `--rosdistro indigo`):
annin_ar4_gazebo: Cannot locate rosdep definition for [gz_ros2_control]
annin_ar4_description: Cannot locate rosdep definition for [xacro]
annin_ar4_moveit_config: Cannot locate rosdep definition for [xacro]
annin_ar4_driver: Cannot locate rosdep definition for [rclcpp_lifecycle]
Continuing to install resolvable dependencies...
#All required rosdeps installed successfully

ycheng517, 2025-02-03:
I think you need to run `source /opt/ros/jazzy/setup.bash` and then repeat the steps. This will set `ROS_DISTRO` which is the first error that you seen.

alonborn, 2025-02-03:
hi, thanks :)
I reinstalled everything (Ubunto, Ros and your driver) and now it works fine, thanks you very much!
Next thing I want to do is to be able to give the robot a position programatically (together with yaw, pitch and roll) and let it move...
will dig into your code and see how this can be done :)
Thanks!
Alon.

closed 2025-02-03 by ycheng517