ycheng517/ar4_ros_driver #44: moveit_servo feature

akshayrmenon, 2025-02-24:
Hi there, Would there be a possibility to have a move_it servo functionality for you robot? It will be very interesting for tele-robotics, I am trying to set it up here; I encounter an issue where somewhere it is blocking my servo topic commands to be send to the trajectory controller.

If there are any ways you could help with the real-time controller, like use the servo_node or any other python api that you would know, please let me know

ycheng517, 2025-02-24:
I do have an example of using moveit_servo to teleoperate the robot. See the annin_ar4_joystick_servo package in https://github.com/ycheng517/ar4_ros_driver_examples . Give it a try and let me know how it works for you!

akshayrmenon, 2025-02-25:
Hi, thank you for your quick response. I am using the mk3 in ros-humble. Not in docker. I tried the servo example: it needed the pilz planning config, which is not part of the humble branch of the ros-driver, but in main. I was not sure whether it needed the pilz planning, so I tried both ways, both which did not work out for me

- commenting out the pilz config side of the joy_servo_example launch 
- Added the pilz planning config to the humble branch of ros driver

If you could tell how I can move forward, that'd be great!

akshayrmenon, 2025-02-25:
Additional update, I have commented our the pilz related launch items; I do see that my commands are being sent to the twist topic from the xbox controller: however it is not being updated to the joint_trajectory_controller.

ycheng517, 2025-02-28:
Hi @akshayrmenon, ROS humble is not supported or tested for the example projects. I quickly tried the joystick servo example and ran into the same issues you mentioned. I suggest you run jazzy in a docker container instead.

akshayrmenon, 2025-02-28:
Ah, alright. I am working on pose tracking with the AR4 using a VIVE tracker, and I needed moveit_servo. Thanks for the update!

ycheng517, 2025-04-06:
Closing issue as it's a known limitation that is not planned to get addressed

closed 2025-04-06 by ycheng517