ycheng517/ar4_ros_driver #46: Adjustable calibration sequence

oskarsVismanis, 2025-02-26:
Hey @ycheng517 !

I've finally had the chance to test on real hardware (everything works great, thanks!). Since we're building a mobile manipulator, we want to add a camera near the end effector and change the gripper fingers to longer ones, made of elastic material. 

While my colleague was creating ways to add these things, he never tried running the arm and I didn't even think this would be an issue, but the AR4 has little room for add-ons (at least in the way it currently calibrates at startup), but he mentioned that the firmware provided by Chris Annin had the option to somehow alter the calibration sequence, which was never ported to the ROS package (I think he meant the ROS 1 package).

Do you know of such a feature in the manufacturer firmware or do you know if such a feature would be possible to add (based on how the ROS 2 firmware is structured currently)? I would be happy to try and implement this, as it's quite a necessity for our use-case (the other options being to add the add-ons after calibration or something else 😆) Maybe it would be useful for your own Tabletop use-case as well?

ycheng517, 2025-02-28:
hey @oskarsVismanis  I believe you're referring to the two-stage calibration process mentioned here: https://youtu.be/F8wv6KpZAZk?si=60YIhoMVP1jMixVw&t=858

It's on my to-do list, I will likely need it soon as well. Similar to the video, I was thinking of making the default calibration 2-stage, first calibration joints 4, 5, 6, and then calibrate joints 1, 2, and 3. Let me know if you're interested in contributing this feature, I'd welcome it!

ycheng517, 2025-03-03:
Awesome, looking forward to your contribution.

oskarsVismanis, 2025-03-13:
Created a pull request for this [here](https://github.com/ycheng517/ar4_ros_driver/pull/47)

ycheng517, 2025-04-06:
PR merged to `humble` and `main` branches, I consider this done!

closed 2025-04-06 by ycheng517