ycheng517/ar4_ros_driver #50: Overrun detected in controller_manager when update rate exceeds 25 Hz

Navaneethakrishnan-S-M, 2025-05-14:
Hi @ycheng517,

I noticed that in the recent commits, the `controller_manager` update rate in `annin_ar4_driver/config/controllers.yaml` was changed from 100 Hz to 60 Hz. While looking into this, I tried increasing the update rate on my end to values above 25 Hz, but I consistently encounter the following warning:

```
[ros2_control_node-1] [WARN] [1747208774.642784562] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 40 Hz. The loop took 28.777818 ms (missed cycles : 2).
[ros2_control_node-1] [WARN] [1747208774.693441324] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 40 Hz. The loop took 29.434481 ms (missed cycles : 2).
```

It seems like anything above 25 Hz causes this overrun issue on my setup.

I’m trying to seek the best way to address this, whether it’s a hardware limitation, some real-time configuration I’m missing, or if the controller settings need tweaking.

I’d really appreciate any guidance you can offer. I’m currently in the middle of a project that depends on this, so if you happen to have any suggestions soon, it would mean a lot!

Thanks again for all the time and effort you’ve put into this package, it's been a huge help so far.

ycheng517, 2025-05-20:
@Navaneethakrishnan-S-M thanks for reporting this issue! I will fix it this week

ycheng517, 2025-05-22:
After some investigation, I suspect that some external factor is slowing down the serial communication to the Arduino nano (when `include_gripper:=True`). I'm still getting overruns when using older commits that definitely did not have any issues, such as 303e9661803ae9e0042f83e0dda97fcceb3aef39. 

Also I don't get overruns when I do `include_gripper:=False`.

@Navaneethakrishnan-S-M can you confirm if this is the case for you too?

Navaneethakrishnan-S-M, 2025-05-26:
After a few more tries, I’m still seeing the overrun warnings occasionally, even with include_gripper:=False. The good news is that it's not flooding the terminal like before, but I do still get occasional messages like these:

```[ros2_control_node-1] [WARN] [1748259816.682937444] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop took 10.538373 ms (missed cycles : 2).
[ros2_control_node-1] [WARN] [1748259817.729608678] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop took 17.209860 ms (missed cycles : 2).
[ros2_control_node-1] [WARN] [1748259819.978292401] [controller_manager]: Overrun detected! The controller manager missed its desired rate of 100 Hz. The loop took 15.893187 ms (missed cycles : 2).```

So it seems the issue is still present, just less frequent. Let me know if there’s anything I should try or tweak further, really appreciate your insights!

ycheng517, 2025-05-31:
@Navaneethakrishnan-S-M if you can try running with a previous commit that did not have this issue, and see this issue exists now, that'd be helpful for me to know!